pycroscopy/pycroscopy #105: Igor ibw translation errors in python 3

ssomnath, 2017-07-18:
pycroscopy\io\translators\igor_ibw.py in _read_parms(ibw_wave)
    173             Dictionary containing parameters
    174         """
--> 175         parm_string = ibw_wave.get('note').decode('utf-8')
    176         parm_string = parm_string.rstrip('\r')
    177         parm_list = parm_string.split('\r')

UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb0 in position 2200: invalid start byte

CompPhysChris, 2018-01-18:
Unfortunately, there doesn't seem to be a standard encoding for ibw files.  I've added a parm_encoding kwarg to the translate method that will allow the user to specify an encoding if it's known.

CompPhysChris, 2018-03-12:
Pull request #133 adds ISO-8859-1 to automatically checked codecs to support older control software.  We still need a good way to determine the encoding or check multiple common encodings.