pycroscopy/pycroscopy #134: upload con conda-forge

carlodri, 2018-03-19:
I was wondering if you were interested in uploading the package on conda-forge, which, as you probably know, is a widespread and robust package managing system. If you agree, I can prepare the feedstock recipe.

ssomnath, 2018-03-19:
Yes, we are indeed interested in and have been planning on making the package available via conda-forge. We just had not gotten around to doing it. It would be great if you could prepare the recipe. Please let us know if you need any information / anything from us.

carlodri, 2018-03-19:
ok @ssomnath. I have started to work on the recipe. I will let you know if I have specific questions. It would also be useful to have the GitHub usernames of the people that would like to be added as recipe maintainers.

ssomnath, 2018-03-19:
Could you please add @CompPhysChris and myself? This should do for now. Also, you should have write access to the package now. Please let me know if you need anything else. Thanks!

carlodri, 2018-03-19:
sure, no problem, happy to help!

carlodri, 2018-03-19:
hmm, looks like the source tarball is not what it should be: it contains a number of nested directories but it doesn't have the structure of a package source. Is this deliberate?

carlodri, 2018-03-23:
I have added the `igor` python package to conda-forge, which was the only missing dependency.

I cannot proceed until #137 is fixed.

carlodri, 2018-03-27:
ok, the conda recipe works fine. @ssomnath @CompPhysChris before releasing on conda-forge I would like to bump version to `0.59.6` in order to fix precedence issues in the actual dev release (`0.59.dev6` precedes `0.59.5`). I'll make a PR for this.

CompPhysChris, 2018-03-28:
Thanks for all your hard work Carlo.  We'd like to invite you to join our slack channel: https://pycroscopy.slack.com/

carlodri, 2018-03-28:
you're welcome, I hope I moved things in the right direction....I'm definitely not a super-duper python expert, particularly on packaging (it can be a nightmare some times....). But I like to try and learn! 

I've joined the slack channel, thanks