pycroscopy/pycroscopy #137: sdist tarball has wrong contents

carlodri, 2018-03-23:
the source tarball available on Pypi is not what it should be: it contains a number of nested directories but it doesn't have the structure of a package source. Is this deliberate?

This is strictly related to #134, since conda-forge is based on `sdist`tarballs.

carlodri, 2018-03-23:
This should be a bug in the TravisCI `deploy` functionality, of which I am no expert. @ssomnath we could try to deploy again bumping the version and adding a new tag to the `master` branch.

CompPhysChris, 2018-03-26:
Ok, I probably should have looked at the source after I set up to have Travis auto-build it.  Turns out I was using bdist instead of sdist.  It should be fixed now,

carlodri, 2018-03-26:
Thanks for stepping in @CompPhysChris 

Well, there are some quirks here:

- 0.59.5d is not a valid PEP440 version
- 0.59.dev6 _precedes_ 0.59.5

Last, the Travis deployment errors out on python 3.6 because you are deploying twice to PyPi (first for 2.7, then for 3.6).  A simple fix would be to move the `python: '3.6'` item under the `on:` group in the `deploy` group (I think it has no effect in the present position).

Eventually, I would bump version to 0.59.6 to fix distribution issues and version numbering.

carlodri, 2018-03-26:
If you want I can do that for you, only I wasn't sure that you wanted a new version to be released yet

CompPhysChris, 2018-03-27:
I'm almost done with a new translator.  I should have it finished later this morning and will bump the version then.

carlodri, 2018-03-27:
@CompPhysChris you might want to review and integrate my changes in #138 

feel free to dispose of the PR once you're done choosing what to keep.

CompPhysChris, 2018-03-27:
I've merged your pull request and tagged for release.

closed 2018-03-27 by CompPhysChris