pycroscopy/pycroscopy #160: NanonisTranslator cannot translate sxm file

donpatrice, 2018-05-31:
I have a sxm file that we would like to read and translate into pycroscopy but this does not work.

```
>>> nt = NanonisTranslator(sxm_file)
>>> nt.translate()
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-240-1db0a86297c7> in <module>()
----> 1 nt.translate()

~/ownCloudFHI/FHI/github.molgen.mpg.de/saasmi/saasmi/.venv/lib/python3.5/site-packages/pycroscopy/io/translators/nanonis.py in translate(self, data_channels, verbose)
     71         """
     72         if self.parm_dict is None or self.data_dict is None:
---> 73             self._read_data(self.data_path)
     74 
     75         if data_channels is None:

~/ownCloudFHI/FHI/github.molgen.mpg.de/saasmi/saasmi/.venv/lib/python3.5/site-packages/pycroscopy/io/translators/nanonis.py in _read_data(self, grid_file_path)
    153 
    154         parm_dict = dict()
--> 155         for key, parm_grid in zip(header_dict['fixed_parameters'] + header_dict['experimental_parameters'],
    156                                   signal_dict['params'].T):
    157             parm_dict[key] = parm_grid

KeyError: 'fixed_parameters'
```
The problem seems to be the missing key 'fixed_parameters' in the header_dict.
```
>>> from pycroscopy.io.translators.df_utils.nanonispy.read import Scan
>>> s = Scan(sxm_file)
>>> s.header.keys()
dict_keys(['scan_pixels', 'current>offset (a)', 'scan_time', 'current>gain', 'z-controller>i gain', 'scan_range', 'data_info', 'current>current (a)', 'z-controller>tiplift (m)', 'z-controller', 'current>calibration (a/v)', 'z-controller>controller name', 'scan_file', 'z-controller>switch off delay (s)', 'z-controller>z (m)', 'z-controller>controller status', 'z-controller>time const (s)', 'rec_time', 'scan_offset', 'z-controller>setpoint', 'rec_temp', 'nanonis_version', 'acq_time', 'scanit_type', 'z-controller>setpoint unit', 'z-controller>p gain', 'bias', 'comment', 'scan_dir', 'scan_angle', 'rec_date'])
```
How can I fix this issue?

Thanks a lot.

ramav87, 2018-05-31:
Thanks for raising this. It looks like because it is .sxm the 'fixed parameters' and 'experimental parameters' ought to be the same. What happens if you simply remove the header_dict['fixed_parameters']  in the zip statement?  You would probably need to check if the dictionary is available though (because it would be for 3ds files).

CFriesn, 2018-05-31:
Hello! Just started looking into pycroscopy and came here to raise this issue. Although trying to piece together my own translator has certainly been educational, a canonical solution would be great!

ssomnath, 2018-05-31:
@CFriesn, we welcome any feedback to improve our code and documentation. Would you mind elaborating? What aspects do you find confusing? I invited you to our slack group to alleviate the communication difficulties.

Also, what data / file format are you interested in translating?

CFriesn, 2018-05-31:
Could I get another invitation to the slack group? I'm a little ashamed to admit I've not used it before, but I've signed up now so hopefully it will work! I will gladly put together some feedback.

For the sake of this thread however, I am trying to translate STM data in the form of Nanonis V4 files, particularly .sxm and .dat files. Right now I have it working for sxm files via nanonispy -> NumpyTranslator as a stand-alone script:

```
from pathlib import Path
import numpy as np
import nanonispy as nano
from pycroscopy.core import Dimension
from pycroscopy.core.io import NumpyTranslator

sxm_path = Path('.\ExampleImage.sxm')
channel = 'Z'
h5_path = Path('.\Test.h5')
unit = 'm'

nanodat = nano.read.Scan(str(sxm_path))

xpx = int(nanodat.header['scan>lines'])
ypx = int(nanodat.header['scan>pixels/line'])

xspan = nanodat.header['scan_range'][0]
yspan = nanodat.header['scan_range'][1]

xdim = Dimension('X','m',np.linspace(0,xspan,xpx))
ydim = Dimension('Y','m',np.linspace(0,yspan,ypx))

dat = nanodat.signals[channel]['forward']

specval = Dimension(channel,unit,1)

trans = NumpyTranslator()
trans.translate(str(h5_path),
                'Raw_Data',
                dat.reshape((ypx*xpx,1)),
                channel,
                unit,
                [ydim, xdim],
                [specval])
```
But clearly having it implemented as a proper class will be better!

ssomnath, 2018-05-31:
@CFriesn I resent the invitations. Perhaps I am sending the invitations to the wrong address (I had to look you up on google)? If your are uncomfortable sharing your email address publicly here, please write to pycroscopy@gmail.com using the address to which I should send out the invitation.

ssomnath, 2018-05-31:
@donpatrice and @CFriesn - If you are interested in making changes to the source code, we invite you to do so. I have given you both write access to the repository. Please feel free to make your own branches >> make changes >> commit changes >> push commits >> then do a pull request. @CompPhysChris or I will then merge changes.

CompPhysChris, 2018-06-01:
I've just created a new branch that has my current work on the sxm support, sxm_dev.  @donpatrice  and @CFriesn, you are both welcome to work off of this branch as well.  I've moved much of the old parameter parsing into the _parse_3ds_parms function, and added a _parse_sxm_parms function.

ssomnath, 2018-06-01:
@donpatrice @CompPhysChris @CFriesn - please communicate details with each other to avoid duplication of efforts