pycroscopy/pycroscopy #163: utf-8 problem with NanonisFile

donpatrice, 2018-05-31:
I have a sxm file, that causes some problems when reading it with NanonisFile:
```
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb0 in position 67: invalid start byte
```
I found that the relevant part differs from [nanonispy](https://github.com/underchemist/nanonispy/blob/master/nanonispy/read.py) by
```
try:
    entry = line.strip().decode()
except UnicodeDecodeError:
    warnings.warn('{} has non-uft-8 characters, replacing them.'.format(f.name))
    entry = line.strip().decode('utf-8', errors='replace')
```
In particular these lines are missing in pycroscopy. Is there a reason for this? Adding these lines solves my problem.

CompPhysChris, 2018-06-01:
Pycroscopy is currently using it's own version of nanonispy.  When we first added it, the package hadn't been updated in years.  Now that it's been updated, we will likely add it to the requirements list and remove our copy of the code.

CompPhysChris, 2018-06-06:
Updating nanonispy files seems to have fixed this.  Closing issue

closed 2018-06-06 by CompPhysChris