pycroscopy/pycroscopy #182: AttributeError in gIV Notebook help wanted

cuu123, 2018-07-03:
The notebook can't run this line：

# Load the raw dataset
h5_path = px.io_utils.uiGetFile('*.h5', 'G-mode IV dataset')

AttributeError: 'module' object has no attribute 'uiGetFile'

Is it possible to use usid instead of px? But Conda install pyUSID seems not working, the message is

PackagesNotFoundError: The following packages are not available from current channels

So how to solve it?

ramav87, 2018-07-03:
When you pip install pycroscopy it should also automatically install pyUSID. The gIV notebook has not been updated since the latest changes were made to px. Anyhow, if you try

import pyUSID as usid 
h5_path = usid.io_utils.file_dialog(file_filter='Translated file (*.h5)')) this should work. 

See the BE processing notebook https://github.com/pycroscopy/pycroscopy/blob/master/jupyter_notebooks/BE_Processing.ipynb for more details.

ssomnath, 2018-07-03:
@cuu123 I updated the notebook. Please check out the notebook again.

ssomnath, 2018-07-03:
Also please consider joining the pycroscopy slack group to interact with the user community. https://pycroscopy.slack.com

closed 2018-07-03 by ssomnath