pycroscopy/pycroscopy #192: Out of date notebooks

DancingQuanta, 2018-10-05:
I am trying to do a tutorial here Tutorial_04_Interactive_Visualization. But I found that some things are broken because the functions in the notebooks have been renamed, or changed functionality.

CompPhysChris, 2018-10-05:
Thanks for bringing this to our attention @DancingQuanta.  I'll take a look at that notebook and get it updated and working again.

DancingQuanta, 2018-10-06:
Great, these notebooks will allow me to make use of such an awesome library that is pycroscopy!

ssomnath, 2018-10-08:
@DancingQuanta - please see [this notebook](https://github.com/pycroscopy/CNMS_UM_2018_SPIMA/blob/master/09_Interactive_Visualizer.ipynb) instead to learn about interactive visualization, which was part of a [recent workshop](https://github.com/pycroscopy/CNMS_UM_2018_SPIMA) on imaging and spectral data analysis.

closed 2018-10-08 by ssomnath