pycroscopy/pycroscopy #204: Importing nanoscope 9.4 files fails

flounderscore, 2019-04-08:
I cannot import a file created with NanoScope 9.4. The error message is:

> File "...translators\bruker_afm.py", line 321, in _read_image_layer
> data_mat = data_vec.reshape(layer_info['Number of lines'], layer_info['Samps/line'])
> ValueError: cannot reshape array of size 524288 into shape (512,512)

The issue appears to be a bug in the NanoScope software >= 9.2 where all data is 4 bytes per pixel even though the header says otherwise.

See line 391 in https://sourceforge.net/p/gwyddion/code/HEAD/tree/trunk/gwyddion/modules/file/nanoscope.c#l31

ramav87, 2019-04-08:
We could probably add a flag to check for this. Is the version of the Nanoscape used to create the file stored?

ssomnath, 2019-04-11:
Pycroscopy will benefit if Gwyddion is somehow integrated into it -  file reading, filters, etc. I know that they have a python package / interface somewhere

ramav87, 2019-05-15:
I have made a temporary fix which is available on my branch. Please let me know if it works for you.

ziatdinovmax, 2019-05-15:
> Pycroscopy will benefit if Gwyddion is somehow integrated into it - file reading, filters, etc. I know that they have a python package / interface somewhere

https://github.com/tuxu/gwyfile

JayBee38, 2019-05-16:
I've been confronted to the same reshape issue, the fix worked for me !