pycroscopy/pycroscopy #205: Error during Importing pycroscopy

ayfliu, 2019-05-22:
I have this error when importing pycroscopy (0.60.3)

~\PyMOL\envs\PyEMMA\lib\site-packages\pycroscopy\io\hdf_writer.py in <module>
     14 import h5py
     15 
---> 16 from pyUSID.io.hdf_utils import assign_group_index, write_simple_attrs, attempt_reg_ref_build, write_region_references
     17 from .virtual_data import VirtualGroup, VirtualDataset, VirtualData
     18 from ..__version__ import version

ImportError: cannot import name 'attempt_reg_ref_build'

I have pyUSID 0.0.6.1 imported in python 3.6.7

ramav87, 2019-05-22:
Yes, this is a known issue that has been fixed on the master. Please clone master until the pypi version is updated.

closed 2019-05-22 by ramav87

ssomnath, 2019-05-24:
You should be able to use pyUSID 0.0.6.2 alternatively.

ayfliu, 2019-05-24:
That was solved by cloning the master. Thanks!