pycroscopy/pycroscopy #207: Igor IBW translator - loss of metadata

JulianeWeb, 2019-06-13:
Hi! 
I am using the Igor IBW translator (IgorIBWTranslator) to convert large amount of AFM data. 
The ibw file has stored a lot of metadata, which normally can be accessed using e.g. gwyddion or the Igor Software. When converting the files into hdf5 using the IgorIBWTranslator, this metadata is lost. There is other metadata added instead. How can I change the code that the metadata (ScanRate, ScanPoints, ScanLines, ScanSize) is added into the hdf5 file? 
I would appreciate help with this a lot, thanks!

ramav87, 2019-06-13:
Hi JulianeWeb, it looks like within the Igor IBW Translator class the function you want to modify is _read_parms() . It appears to cycle through the parameters in parm_list which is itself from ibw_wave.get('note') . Take a look at the output for that, and for ibw_wave.get('wave_header') . Do these have the required metadata?

JulianeWeb, 2019-06-14:
Hi ramav87, thanks for your answer. I am trying to see the output for the parm_list but it does not really work for me at the moment. I am working with Liam Collins on an AFM project, if there is any contact possibility, besides github, I could contact you and describe the issue more in detail. I can also send you some example data. 
Thank you!

ramav87, 2019-06-15:
Certainly, you can stop by my office next week when you are at CNMS (e.g.,
on Wednesday).

Rama

ramav87, 2019-06-24:
Closing issue - the attributes including scan rate, scan points, etc. are in the measurement group.

closed 2019-06-24 by ramav87