pycroscopy/pycroscopy #243: Gallery examples are broken

stefanv, 2021-01-16:
Several examples on the gallery page failed to execute: https://pycroscopy.github.io/pycroscopy/auto_examples/index.html

ssomnath, 2021-01-18:
Hi @stefanv. Nice to hear from you. 

Indeed, multiple of our examples have failed to execute. We are in the process of performing a major overhaul of the pycroscopy package that will address this issue and others. 

It may take a tad bit longer than usual to fix these bugs since 
* we are currently migrating away from our previous CI/CD provider to newer platform. Both failing examples actually work locally with the latest versions of the dependency packages but we we are having to manually build and deploy until we get our new CI/CD pipelines running.
* we are in the middle of a major restructuring effort that will redefine the pycroscopy package.
  * The pycroscopy package is currently being broken down into more focused packages that better suit the end user (even though this means additional administrative and management efforts on our end) and 
  * we have a new [data model](https://pycroscopy.github.io/sidpy/notebooks/00_basic_usage/create_dataset.html) and a new [data storage schema](https://pycroscopy.github.io/pyNSID/nsid.html#strawman-nsid-specifications) that will underpin most of the pycroscopy ecosystem of packages.  

Rest assured, we are working towards addressing issues as best we can.

stefanv, 2021-01-18:
@ssomnath Thanks for the update—I look forward to seeing the new pycroscopy ecosystem documentation online soon.  Best of luck with the ambitious refactor!

ssomnath, 2021-03-05:
Pycroscopy is being completely refactored. Please see #245. The reimagined pycroscopy is being put together in the [phoenix](https://github.com/pycroscopy/pycroscopy/tree/phoenix) branch. We will be swapping out the documentation as well in the process.

ssomnath, 2021-04-16:
Examples were removed since they will not be relevant to the reimagined pycroscopy and documentation has now been updated

closed 2021-04-16 by ssomnath