pycroscopy/pycroscopy #244: PiFM Translator

ramav87, 2021-02-12:
Of particular importance for @rajgiriUW  and @kongjy , we are migrating basically all translators out of pycroscopy into their respective packages (most of them go to SciFiReaders). Please advise as to how we should handle the pifm translator. One option is to move it in with the other USID-based translators, which are now in BGLib.

rajgiriUW, 2021-02-12:
Jess is no longer involved in PiFM so I don't think she has a strong opinion here. But, regarding this case, I think moving the SciFiReaders makes sense, in which case I'll need to update it as I haven't had to use it in awhile.

ramav87, 2021-02-12:
Thanks! It can be ported to bglib without change, but in scifi readers it
will need to be changed to sidpy datasets. The change shouldn't take long,
though.

kongjy, 2021-02-12:
I agree that it makes sense to move it to SciFiText given the specific scope of BGLib. Thanks for volunteering to do it, @rajgiriUW. Do let me know if I can help in anyway.

rajgiriUW, 2021-09-14:
I've written a roughly working version in SciFiReaders that converts to sidpy Datasets and then writes via pyNSID to HDF5. 
https://github.com/pycroscopy/SciFiReaders/blob/raj-pifm/SciFiReaders/readers/microscopy/spm/afm/pifm.py

I'll submit a PR when it's validated on a few more sets. Right now having trouble getting the laser power spectrum to write correctly

rajgiriUW, 2021-09-14:
The PiFM translator is still in pycroscopy. I don't mind where it moves, though BGLib is fine since Phoenix reorg is still ongoing for Pycroscopy

ramav87, 2021-09-14:
We can certainly move it to BGLib for the time being. Would you like me to do so?

rajgiriUW, 2021-09-15:
Sure, I think that's fine to move there for now-- it doesn't make too much sense there but I guess the idea is hopefully people use the SciFiReaders one once I check it out a little further.