pycroscopy/pycroscopy #251: ModuleNotFoundError: No module named 'pyUSID.io.write_utils'

luv2python, 2021-04-04:
Hello,

My name is Sam. I work at Intel. Since I often analyze AFM data and SEM data and I am a python programmer, I am interested in using pycroscopy. I was able to install it successfully on my Windows 10 laptop. However, when I import pycroscopy, I got the following error. Could you please help me to resolve it?

Thanks!

Sam

 Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import pycroscopy as px
  File "C:\Users\samxu\AppData\Roaming\Python\Python38\site-packages\pycroscopy\__init__.py", line 10, in <module>
    from .io import translators
  File "C:\Users\samxu\AppData\Roaming\Python\Python38\site-packages\pycroscopy\io\__init__.py", line 14, in <module>
    from . import translators
  File "C:\Users\samxu\AppData\Roaming\Python\Python38\site-packages\pycroscopy\io\translators\__init__.py", line 10, in <module>
    from .be_odf import BEodfTranslator
  File "C:\Users\samxu\AppData\Roaming\Python\Python38\site-packages\pycroscopy\io\translators\be_odf.py", line 18, in <module>
    from .df_utils.be_utils import trimUDVS, getSpectroscopicParmLabel, parmsToDict, generatePlotGroups, \
  File "C:\Users\samxu\AppData\Roaming\Python\Python38\site-packages\pycroscopy\io\translators\df_utils\__init__.py", line 1, in <module>
    from . import be_utils
  File "C:\Users\samxu\AppData\Roaming\Python\Python38\site-packages\pycroscopy\io\translators\df_utils\be_utils.py", line 24, in <module>
    from pyUSID.io.write_utils import create_spec_inds_from_vals, Dimension
ModuleNotFoundError: No module named 'pyUSID.io.write_utils'

rajgiriUW, 2021-04-04:
Hi Sam,
The issue is there have been a number of changes in the interim for pyUSID and related, so that particular import doesn't work. That being said, if you know how to pull from source, then the latest commit on Pycroscopy _should_ work last time I checked. I believe @ramav87 and others are still finishing import fixes before pushing a new minor release. I don't believe be_utils is even in Pycroscopy anymore in the latest commit.

Otherwise, it might help to know what you are using Pycroscopy for. At this point, it's sort of a grabbag of functions, so depending on if the use is on the functionality within Pycroscopy specifically (some particular analysis, say) it might be unnecessary as opposed to using pyNSID/sidpy directly for converting your SEM data into Python.

Best,
Raj

luv2python, 2021-04-04:
Hi Raj,

Thanks for reply to me so quick! You asked a good question. The reason I am interested in pycroscopy is that I am trying to write my own python code to analyze images from SEM (scanning electron microscopy) and AFM (atomic force microscopy). While both tools came with software for analyzing the width of the feature for SEM and height of the feature for AFM, there are limitations. By writing my own python code, I have the freedom to break such limitations.  They both are 2D images with grey scale for SEM and color map for AFM. So, I think there is a way like MatLab does to extract information with some functions. Can pycroscopy do something like that?

Thanks!

Sam

ramav87, 2021-04-05:
Hi Sam,

Pycroscopy at the moment is under a great deal of development and will
incorporate functions for image analysis of SEM, AFM, and other such
images. At the time pycroscopy was developed, much of it was highly
specialized towards advanced AFM spectroscopy modes, and thus probably has
little utility for generic 2D image analysis in its current state. That
said, there are a few other packages within the pycroscopy ecosystem (
github.com/pycroscopy) that are more suited towards image analysis,
particularly if you are working with TEM images (pyTEMLib) or want to
dabble with some deep learning (atomAI). For circumventing the error, you
can uninstall pycroscopy and pyUSID and scikit-image and then install them
in this order:

!pip install pycroscopy
!pip install scikit-image==0.14.2
!pip install pyUSID==0.0.8.1

I found this combination works on Google colab. That said, as mentioned
above it's not clear the older pycroscopy has what you need, so we
recommend to look at the other packages, or wait for a few months until the
phoenix branch of pycroscopy (
https://github.com/pycroscopy/pycroscopy/tree/phoenix) becomes the main
branch.

Kind Regards,
Rama

rajgiriUW, 2021-04-05:
Thanks Rama! Just to add, if you look under io/translators there are a few available for different instruments. However, what you probably want are the file readers in another package, [SciFiReaders](https://github.com/pycroscopy/SciFiReaders). That's if your goal as a first step is to just get the data into Python. The PyNSID library then provides some convenient functionality for handling that data.

Best,
Raj

luv2python, 2021-04-05:
Hi Rama,

Thank you so much for the detail explanations and advice! I will give it a try. Looks like you guys are doing some great works. Looking forward to your new release!

Regards,

Sam

ssomnath, 2021-04-07:
Chad Parish brought up this issue as well. We do need a fix. I know what I'll be working on this week's hackathon :-)

ssomnath, 2021-04-16:
This issue has been fixed in the latest version of pycroscopy - [0.60.8](https://github.com/pycroscopy/pycroscopy/releases/tag/0.60.8)

closed 2021-04-16 by ssomnath