pycroscopy/pycroscopy #252: Installed pycroscopy

darssita2017, 2021-04-15:
I installed jupyter and this software but not knowing how to start code writing for my KPFM image analysis. Please help me out in this matter

ssomnath, 2021-04-16:
@darssita2017 - Would you mind elaborating what you are interested in doing with KPFM images?

darssita2017, 2021-04-17:
Dear Sir,
I want to see contact potential variation. And to draw figure for article.
Thank you very much sir for your time and consideration.
Kind regards
Aniruddh

darssita2017, 2021-04-17:
Dear sir,
I want to analyze these KPFM images. From these KPFM images and I would
like to calculate the work function and charge carriers. Please find the
files.  The data size is large so I am sending two-three images.
yours sincerely,

Dr. Aniruddh Bahadur Yadav

Velagapudi Ramakrishna Siddhartha Engineering College
(Autonomous) Kanuru , Vijayawada
Andhra Pradesh - 520007,INDIA

darssita2017, 2021-04-29:
Dear sir,
I am eagerly waiting for a reply from your side.
yours sincerely,

Dr. Aniruddh Bahadur Yadav

Velagapudi Ramakrishna Siddhartha Engineering College
(Autonomous) Kanuru , Vijayawada
Andhra Pradesh - 520007,INDIA

ramav87, 2021-04-29:
Hi  Aniruddh,

Sorry for the delay. It certainly seems that your work can be more readily
accomplished by use of available software packages for 2D AFM image
analysis, such as Gwyddion. Have you explored this option?

Pycroscopy in its original form was more geared towards higher dimensional
analysis, for spectral unmixing and so forth. While there will be more 2D
image analytics incorporated, it will take a few months before we are ready
to debut the new iteration of the package.

Kind Regards,
Rama

darssita2017, 2021-04-30:
Dear Sir,
I  used Gwyddion for analysis of my images. I thought your software can
give better insight and many file format are not open in Gwyddion. Hope you
may understand my problem.
Yours sincerely,

ssomnath, 2021-04-30:
@darssita2017 - As @ramav87 says, we may indeed have kpfm analyses capabilities in the future within the pycroscopy ecosystem. However, this may take more time than you are willing accommodate. If Gwyddion indeed does not serve your needs, you are welcome to use the frameworks in pycroscopy to develop solutions for your problems. The first step would be for you to develop a sidpy.Reader to open your data files. Instructions are available in the documentation for sidpy. Once you are able to read your data and metadata, we can guide you in extracting insights.

darssita2017, 2021-04-30:
Dear sir,
Yes sir I will open it and learn to open the image in sidpy.Reader. Once
get the image send you for further analysis help.
yours sincerely,

Dr. Aniruddh Bahadur Yadav

Velagapudi Ramakrishna Siddhartha Engineering College
(Autonomous) Kanuru , Vijayawada
Andhra Pradesh - 520007,INDIA

darssita2017, 2021-04-30:
Dear sir,
I am a beginner in python so unable to open my file using your codes given
on this link.
[image: image.png]
The website I am using
https://pycroscopy.github.io/sidpy/notebooks/00_basic_usage/create_dataset.html
.
please tell me what should I do?
yours sincerely,

Dr. Aniruddh Bahadur Yadav

Velagapudi Ramakrishna Siddhartha Engineering College
(Autonomous) Kanuru , Vijayawada
Andhra Pradesh - 520007,INDIA

ssomnath, 2021-04-30:
@darssita2017 - Which microscope did you use to do the KPFM measurements? Have you tried to export the data to a format that Gwyddion can understand?

darssita2017, 2021-04-30:
> @darssita2017 <https://github.com/darssita2017> - Using pycroscopy, or
> most python packages for that matter, requires understanding of python,
> basic programming concepts, and scripting since such packages do not come
> with a graphical user interface that most users are familiar and
> comfortable with. As it stands right now, the pycroscopy ecosystem does not
> have all the capabilities you are looking for. We are a small team of
> volunteers who write the code and it therefore takes time to add more
> features, such as the ability to open the data files you speak of. We are
> currently directing all efforts towards fleshing out the core pycroscopy
> package. We are happy to direct you towards resources you could use to help
> yourself in the meantime. I would suggest that you learn python
> programming, then scientific python (numpy, scipy, dask, etc.), then about
> our ecosystem (sidpy, ScopeReaders, pyNSID, and pycroscopy), and then start
> using our frameworks and tools to develop your own codes to read your data
> from the files. Once you are comfortable with these basics, we can guide
> you so that you may develop your own KPFM data analysis codes.

Dear sir,
Definitely, I learn python. I started to and completed some online courses
on python. Please tell me some references elaborating much and help me in
writing code.
your sincerely,

Dr. Aniruddh Bahadur Yadav

Velagapudi Ramakrishna Siddhartha Engineering College
(Autonomous) Kanuru , Vijayawada
Andhra Pradesh - 520007,INDIA

ssomnath, 2021-04-30:
@darssita2017 - I agree with @ramav87 that the quickest route to getting the insights / information you need is to request your collaborators in IIT Bombay to export to a format (preferably csv / text instead of a color image) that provides you access to the data. 

I am not aware of the SPM model you referred to and am unable to provide further guidance. 

You can certainly find numerous tutorials online to learn scientific python. We do not teach or provide documentation on these topics. However, you may find [this page](https://pycroscopy.github.io/sidpy/external_guides.html) useful.

darssita2017, 2021-04-30:
> They should be able to export the data to some standard like .txt file
> format though, right?
>
> Rama

Dear sir,
Sir, the measurement was done 2 years ago. I don't, think they have data in
their repository. Other than the jpeg image I have the following files.
kinds regards,

Dr. Aniruddh Bahadur Yadav

Velagapudi Ramakrishna Siddhartha Engineering College
(Autonomous) Kanuru , Vijayawada
Andhra Pradesh - 520007,INDIA

X100_ZnO0002NapPotentialRetrace
RoughnessData:
Parameter                    Full Image Values:  Mask Image Values:  
---------------------------------------------------------------------
Number Of Points:            65536                                   
Average:                     383.049 mV          nan V               
Standard Deviation:          4.325 mV            nan V               
Max:                         394.082 mV          nan V               
Min:                         372.573 mV          nan V               
RMS:                         383.073 mV          nan V               
Average Deviation:           3.794 mV            nan V               
Skew:                        -0.00424            nan                 
Kurtosis:                    -1.18               nan                 
Percent XY:                  100 %               0 %                 
Surface Area:                nan nm²             nan µm²             
Area Percent:                nan %               nan %               
Volume                       nan µm³             nan µm³             

X100_Ti__2_NapPotentialRetrace
RoughnessData:
Parameter                    Full Image Values:  Mask Image Values:  
---------------------------------------------------------------------
Number Of Points:            65536                                   
Average:                     737.638 mV          nan V               
Standard Deviation:          4.806 mV            nan V               
Max:                         750.026 mV          nan V               
Min:                         726.062 mV          nan V               
RMS:                         737.653 mV          nan V               
Average Deviation:           3.994 mV            nan V               
Skew:                        0.0688              nan                 
Kurtosis:                    -0.822              nan                 
Percent XY:                  100 %               0 %                 
Surface Area:                nan nm²             nan µm²             
Area Percent:                nan %               nan %               
Volume                       nan µm³             nan µm³             

X100_ZnO0001NapPotentialRetrace
RoughnessData:
Parameter                    Full Image Values:  Mask Image Values:  
---------------------------------------------------------------------
Number Of Points:            65536                                   
Average:                     398.403 mV          nan V               
Standard Deviation:          7.851 mV            nan V               
Max:                         420.854 mV          nan V               
Min:                         382.788 mV          nan V               
RMS:                         398.481 mV          nan V               
Average Deviation:           6.546 mV            nan V               
Skew:                        0.764               nan                 
Kurtosis:                    -0.475              nan                 
Percent XY:                  100 %               0 %                 
Surface Area:                nan nm²             nan µm²             
Area Percent:                nan %               nan %               
Volume                       nan µm³             nan µm³             

X100_pt0000NapPotentialRetrace
RoughnessData:
Parameter                    Full Image Values:  Mask Image Values:  
---------------------------------------------------------------------
Number Of Points:            65536                                   
[... 23 lines cut ...]
Average Deviation:           8.142 mV            nan V               
Skew:                        -0.869              nan                 
Kurtosis:                    -0.269              nan                 
Percent XY:                  100 %               0 %                 
Surface Area:                nan nm²             nan µm²             
Area Percent:                nan %               nan %               
Volume                       nan µm³             nan µm³             

SKPM_0001NapPotentialRetrace
RoughnessData:
Parameter                    Full Image Values:  Mask Image Values:  
---------------------------------------------------------------------
Number Of Points:            65536                                   
Average:                     762.714 mV          nan V               
Standard Deviation:          14.675 mV           nan V               
Max:                         811.543 mV          nan V               
Min:                         738.783 mV          nan V               
RMS:                         762.855 mV          nan V               
Average Deviation:           12.584 mV           nan V               
Skew:                        0.482               nan                 
Kurtosis:                    -0.74               nan                 
Percent XY:                  100 %               0 %                 
Surface Area:                nan nm²             nan µm²             
Area Percent:                nan %               nan %               
Volume                       nan µm³             nan µm³

rajgiriUW, 2021-04-30:
Do you have the original raw data (the file saved by the Pico)?

darssita2017, 2021-04-30:
Dear sir,
In the file attached here I have only this data.
kind regards,

Dr. Aniruddh Bahadur Yadav

Velagapudi Ramakrishna Siddhartha Engineering College
(Autonomous) Kanuru , Vijayawada
Andhra Pradesh - 520007,INDIA

ramav87, 2021-04-30:
Can you please attach a link to the file? We don't see any jpg file or which file the text you sent comes from.

closed 2021-04-30 by ramav87

darssita2017, 2022-10-25:
Dear Sir,
Please circulate the information among your known.
best regards,

Dr. Aniruddh Bahadur Yadav

Sr. Assistant Professor

Department of Electronics Communication Engineering

Velagapudi Ramakrishna Siddhartha Engineering College
(Autonomous) Kanuru , Vijayawada
Andhra Pradesh - 520007,INDIA