pycroscopy/pycroscopy #254: Web applications

ssomnath, 2021-05-04:
Consider using [streamlit](https://streamlit.io) for deploying simple and elegant web applications. One example use case is @gduscher's idea of providing a quick visualizer of all data contained in a folder. Unlike a jupyter notebook, this might be a compact and tamperproof method for deploying novice or student focused applications. 

There is one other package similar to ``Streamlit``. I will post it here once I manage to remember it.

rajgiriUW, 2021-05-05:
This is a good idea! I've been exploring Streamlit for some other work in our lab. There are some things to keep in mind with this idea, depending on the exact idea, one of which is that in my (limited) Streamlit experience the File I/O component is a little underpowered. But an app that walks through data locally hosted within the relevant repository might obviate that, and help show some specifics.

ssomnath, 2021-05-05:
Great to hear positive feedback about the Idea and streamlit, @rajgiriUW. I could swear I came across another package similar to streamlit but I can't seem to track it down. Please do let us know if you are aware of alternatives.

ssomnath, 2021-05-11:
Found the other package that is similar to streamlit - [Pweave](http://mpastell.com/pweave/index.html)

ziatdinovmax, 2021-05-12:
> Found the other package that is similar to streamlit - [Pweave](http://mpastell.com/pweave/index.html)

Looks like their last commit to master branch was 2 years ago. I wonder if this project is still active?
[image]

ssomnath, 2021-05-12:
@ziatdinovmax Thanks for catching that. On second glance, Pweave appears to be an alternative to LaTeX rather than Jupyter notebooks or Streamlit.

ziatdinovmax, 2021-05-22:
It looks like Streamlit is also one of the "officially" [recommended ways](https://github.com/paperswithcode/releasing-research-code#making-demos-tutorials-executable-papers) to make examples/tutorials for ML research papers:
[image]

ramav87, 2021-09-10:
@rajgiriUW Can you possibly update us on the use of streamlit for web apps? We are looking for some way to utilize the interactive elements of Matplotlib in a web app, but streamlit only supports the % inline (static) versions. I have looked into other solutions, but they all have challenges.

rajgiriUW, 2021-09-10:
I only used it for a web app version to process transistor data in our lab. I think there is a separate streamlit plotting method that allows more dynamic figures (like matplotlib interactivity) but haven't looked really into it. I can take a look.