pycroscopy/pycroscopy #258: Dimension Error with Python 3.8

Liambcollins, 2021-07-12:
@ssomnath I got the following error when using Dimension (which broke the translators)

I was using Python 3.8.3 (and imported pyUSID 0.10.0 from master). Reverting back to 3.7 appears to have fixed the issue. Just wanted to give a heads up.

[image]

ramav87, 2021-07-12:
Thanks Liam. I don't seem to be able to find much on the python website about this, but this is the only thing that mentions __new__()

"A RuntimeError is now raised when the custom metaclass doesn’t provide the __classcell__ entry in the namespace passed to type.__new__. A DeprecationWarning was emitted in Python 3.6–3.7. (Contributed by Serhiy Storchaka in bpo-23722.)"

I can't quite decipher this though. Thoughts, @ssomnath , @ziatdinovmax , @gduscher ?

rajgiriUW, 2021-07-13:
Hi Liam,
This is odd to me because I use 3.8.5 and haven't had this issue come up. I tried that line you posted above and it worked fine. Can you try creating a new environment and seeing if the new issue pops up? Can you confirm the pyUSID version, though I don't think that's the issue? 

[image]

Raj

Liambcollins, 2021-07-14:
Hey Raj, you are right I reinstalled python 3.8.5 and all looks good. I was using 0.10.0.. I dont know what version of sidpy I was running, maybe that could have been it?

Liam

closed 2023-07-03 by ramav87