pycroscopy/pycroscopy #260: Archiving master

ramav87, 2021-09-17:
I am now wanting to make phoenix the master branch in px. However, it seems that master is a few commits ahead of legacy. I am thinking of merging master with legacy. If there are any issues, please let me know @ssomnath

ssomnath, 2021-09-18:
@ramav87 - It is fine to make phoenix the primary branch. My suspicion is that master largely has had CI/CD and documentation commits since phoenix began. You can merge those commits into phoenix. That way, phoenix has all the documentation updated from master. 

I see that you want to release an alpha version of phoenix. If this is the case, you could even merge master with phoenix. The CI/CD pipeline should automatically create a new release on pypi. 

We should not be touching legacy at all. That is the branch someone could use if they want to stick to the old pyUSID based functions. If I am not mistaken, that branch has some old translators that have not been transitioned to SciFiReaders. So, there is value to keeping this branch as is.

ramav87, 2021-09-18:
Thanks Suhas. Yes, I do not wish to update legacy, however I was surprised to find master ahead of legacy in some commits and also that legacy was ahead in some commits as compared to master. As it stands, legacy will no longer be touched. I am perfectly happy with users installing the older version of px directly from PyPi (which is what seems to be happening).