pycroscopy/pycroscopy #263: Could not import pycroscopy

slps9061008, 2022-02-01:
Hi 

I am really new here and right now I have issues when I tried to import asylum files, could you please help me out?

First I tried to import pycroscopy following the instruction here: https://groups.google.com/g/pycroscopy/c/5BzXC6Mbb6I
When I used the command import pycroscopy it shows the following message:
 
```
File "C:\ProgramData\Anaconda3\lib\site-packages\skimage\util\arraycrop.py", line 8, in <module>
   from numpy.lib.arraypad import _validate_lengths

ImportError: cannot import name '_validate_lengths'

```
I searched online and it seems like that pycroscopy it trying to call some function from old numpy, I tried to switch to old numpy but then other parts have issues too. Do we already have newer version to solve this?

My numpy version is 1.19.5

ramav87, 2022-02-01:
Hi, are you trying to import Igor ibw files? If so you should use SciFiReaders which is a different package within the pycroscopy ecosystem.

slps9061008, 2022-02-01:
Hi yes, and there is a different problem with that....

I used the following code, but it could not popup interactive window in spyder, do you know how it might help?

: %pylab notebook
%gui qt

import sys
from sidpy.io.interface_utils import openfile_dialog_QT
sys.path.insert(0,'../../')
from SciFiReaders import DM3Reader
import SciFiReaders
print('SciFiReaders version: ', SciFiReaders.__version__)
import sidpy
print('sidpy version: ', sidpy.__version__)

ramav87, 2022-02-01:
What is the specific error? Also, please ignore the sidpy file dialog box for now, let's come back to that one.

First, can the file be read by the IgorIBW reader?

slps9061008, 2022-02-01:
Yes, I made it work on Jupiter notebook.

%pylab notebook
%gui qt
import os
import sys
from sidpy.io.interface_utils import openfile_dialog_QT
sys.path.append('../')
from SciFiReaders import IgorIBWReader
import SciFiReaders
print('SciFiReaders version: ', SciFiReaders.version)
import sidpy
print('sidpy version: ', sidpy.version)

ramav87, 2023-07-03:
This looks like an issue with scikit-image. It appears to be solved here https://stackoverflow.com/questions/54241226/importerror-cannot-import-name-validate-lengths

Closing issue.

closed 2023-07-03 by ramav87