pycroscopy/pycroscopy #273: visualize two channels of my *.h5 AFM file as a single 3D map

mtenorio1995, 2022-09-23:
Hi!
I have an Asylum Research *.h5 file with two channels: "Raw (m)", which is channel my_data[0]; "Defl (m)", or my_data[1].
Each channel is a 3D array of dimensions (32, 32, 961):

sidpy.Dataset of type SPECTRAL_IMAGE with:
 dask.array<array, shape=(32, 32, 961), dtype=float64, chunksize=(32, 32, 961), chunktype=numpy.ndarray>
 data contains: Raw (m)
 and Dimensions: 
x:  x (m) of size (32,)
y:  y (m) of size (32,)
Time:  z (s) of size (961,)

 This is how it looks like when plotting only the channel "Defl (m)":
[image]

I want to combine them to have a 4d array: (32, 32, 2, 961) and plot it.

Thanks in advance!

ramav87, 2022-09-23:
Hi! For now we are only supporting the 3D visualizer for line plots and 4D visualizers for images. I will try to create a new visualizer for your case, and get back to you.

mtenorio1995, 2022-09-28:
> Hi! For now we are only supporting the 3D visualizer for line plots and 4D visualizers for images. I will try to create a new visualizer for your case, and get back to you.

Ok! Thanks in advance!

ramav87, 2022-10-04:
The latest sidpy (main branch) now supports multiple channel spectral plotting. This requires you to define a specific 'channel' dimension type when you create the sidpy dataset. Here is an example with synthetic data. 

Closing this issue now. Thanks!

```
def return_quad(x, *parms):
    a,b,c, = parms
    return a*x**2 + b*x + c

data_mat = np.zeros(shape=(10,10,32,2)) #sample dataset with 2 channels
xvec = np.linspace(0,5,32)

#Make some fake data
for row in range(data_mat.shape[0]):
    for col in range(data_mat.shape[1]):
        if row**2 + col**2 <data_mat.shape[0]*data_mat.shape[1]/3:
            a = np.random.normal(loc=3.0, scale = 0.4)
            b = np.random.normal(loc=1.0, scale = 2.4)
            c = np.random.normal(loc=0.5, scale = 1.0)    
        else:
            a = np.random.normal(loc=1.0, scale = 0.8)
            b = np.random.normal(loc=0.0, scale = 1.4)
            c = np.random.normal(loc=-0.5, scale = 1.3)
        data_mat[row,col,:,0] = return_quad(xvec, *[a,b,c]) + 2.5*np.random.normal(size=len(xvec))
        data_mat[row,col,:,1] = -1*return_quad(xvec, *[-1*a,2*b,3*c]) + 1.5*np.random.normal(size=len(xvec))

parms_dict = {'info_1': np.linspace(0,1,100), 'instrument': 'perseverence rover AFM'}

#Let's convert it to a sidpy dataset

#Specify dimensions
x_dim = np.linspace(0, 1E-6,
                    data_mat.shape[0])
y_dim = np.linspace(0, 1E-6,
                    data_mat.shape[1])
z_dim = xvec

channel_dim = [0,1]

#Make a sidpy dataset
data_set = sid.Dataset.from_array(data_mat, title='Current_spectral_map')
    
#Set the data type
data_set.data_type = sid.DataType.SPECTRAL_IMAGE

# Add quantity and units
data_set.units = 'nA'
data_set.quantity = 'Current'

# Add dimension info
data_set.set_dimension(0, sid.Dimension(x_dim,
                                        name='x',
                                        units='m', quantity='x',
                                        dimension_type='spatial'))
data_set.set_dimension(1, sid.Dimension(y_dim,
                                        name='y',
                                        units='m', quantity='y',
                                        dimension_type='spatial'))

data_set.set_dimension(2, sid.Dimension(z_dim,
                                        name='Voltage',
                                        units='V', quantity='Voltage',
                                        dimension_type='spectral'))

data_set.set_dimension(3, sid.Dimension(channel_dim,
                                        name='Channel',
                                        units='a.u.', quantity='Channel',
                                        dimension_type='channel'))

# append metadata            
data_set.metadata = parms_dict
```

closed 2022-10-04 by ramav87